equinor/flotilla #1254: Log authentication attempts

tsundvoll, 2023-12-12:
**Describe the improvement you would like to see**
Log authentication attempts when a user tries to authenticate towards Flotilla

**How will this change existing functionality?**
No functional change for the user. Operations team will get better overview of usage.

**How will this improvement affect the current Threat Model?**
Increases overview of the usage of the application

eivindsjovold, 2023-12-18:
We log authentication attempts in the respective managed identities(linked to app registration) in Azure AD. There should be some setting to bundle this logging with the AI, I propose we investigate this pattern.

aestene, 2024-02-17:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2024-03-14:
> We log authentication attempts in the respective managed identities(linked to app registration) in Azure AD. There should be some setting to bundle this logging with the AI, I propose we investigate this pattern.

Great, I was not aware. This seems sufficient for now. I dont think we need or want to also log this in Application Insights, so I will close this issue.

What we may want in the future is some automatic report / notification if there is a lot of unknown / unsuccessful authentication attempts in a short period of time

closed 2024-03-14 by tsundvoll