equinor/flotilla #1261: Use common function for updating robot properties

aestene, 2023-12-14:
**Describe the improvement you would like to see**
The current implementation of updating individual robot properties using `ExecuteUpdateAsync` had a lot of duplicated functionality throughout the file [RobotService.cs](https://github.com/equinor/flotilla/blob/main/backend/api/Services/RobotService.cs).

This should instead be transformed into one common function which the different specific functions call. The suggested solution is as follows
``` c#
public async Task<Robot?> UpdateRobotStatus(string robotId, RobotStatus status) { return await UpdateRobotProperty(robotId, robot => robot.Status, status) }

private async Task<Robot> UpdateRobotProperty<TProperty>(string robotId, Func<Robot, TProperty> property, TProperty value)
{
    var robotQuery = context.Robots.Where(robot => robot.Id == robotId).Include(robot => robot.CurrentInstallation);
    var robot = await robotQuery.FirstOrDefaultAsync();

    await VerifyThatUserIsAuthorizedToUpdateDataForInstallation(robot!.CurrentInstallation);

    await robotQuery.ExecuteUpdateAsync(robots => robots.SetProperty(property, value));

    robot = await robotQuery.FirstOrDefaultAsync();
    NotifySignalROfUpdatedRobot(robot!, robot!.CurrentInstallation!);

    return robot;
}

However, this solution is currently not supported by Entity Framework (https://github.com/dotnet/efcore/issues/31349) which is why we have to include so much duplicated code. 

They are considering implementing a solution and this issue may be completed when (and if) they complete this issue (https://github.com/dotnet/efcore/issues/32018)
```

**How will this change existing functionality?**
Improve readability and reduce lines of code.

**How will this improvement affect the current Threat Model?**
N/A

aestene, 2024-02-13:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2024-09-03:
This is coming in .NET 10.0 https://github.com/dotnet/efcore/issues/32018#issuecomment-2174348293

tsundvoll, 2025-03-03:
We do this differently now anyways, closing as not planned

closed 2025-03-03 by tsundvoll