equinor/flotilla #1264: Introduce global ordering of mission runs

andchiind, 2023-12-14:
**Describe the improvement you would like to see**
Each new mission run should be assigned an incremented integer.

**How will this change existing functionality?**
This needs to be done with a mutex. This number can then be used to order the mission in the mission queue on the backend. By sending this integer to the frontend we can then have a consistent ordering. Missions that need to run immediately should have the value set to zero as a special case. This integer also allows us to implement reordering of missions in the queue, though this also requires the use of mutexes. Using a large integer accomodates a very large number of missions without running out, but if this is something we're worried about (an unsigned long allows 18,446,744,073,709,551,615 mission runs) then we can allow the value to wrap. To find what the next integer should be the backend should store the counter in a database table. Updating it needs to be done with mutexes (both for reading the last value and writing the new value).

I am open to discussions on this topic. I understand it introduces more overhead and complexity, but I think in the long run we should have a solution for consistent ordering.

**How will this improvement affect the current Threat Model?**
N/A

tsundvoll, 2024-01-22:
Notes:
Plan to add variable on MissionRun
Semaphore in scoped service will not work as intended (needs to be in a hosted service)

UsamaEquinorAFK, 2024-01-24:
https://github.com/equinor/flotilla/pull/1354

UsamaEquinorAFK, 2024-01-26:
https://learn.microsoft.com/en-us/ef/core/modeling/generated-properties?tabs=data-annotations

UsamaEquinorAFK, 2024-01-26:
https://stackoverflow.com/questions/36155429/auto-increment-on-partial-primary-key-with-entity-framework-core

UsamaEquinorAFK, 2024-01-30:
https://github.com/orgs/equinor/projects/85/views/4?pane=issue&itemId=45945484

closed 2024-02-05 by UsamaEquinorAFK