equinor/flotilla #1267: Add link to mission definition in mission run page

andchiind, 2023-12-15:
**Describe the new feature you would like to see**
We should be able to go from the mission run page to the mission definition page with a link.

**Describe the solution you'd like**
The thing I wrote above.

**How will this feature affect the current Threat Model?**
N/A, it would be impressive to make this a security risk

aestene, 2024-02-14:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2025-02-13:
This should be on wait until the mission definition page has been redone. Ideally the mission definition page should also list the previous runs for that mission definition in my opinion.