equinor/flotilla #1291: The frontend should receive a warning the robot could not be returned home

Afonso-2403, 2023-12-22:
**Describe the new feature you would like to see**
When the scheduling of a return home mission fails in [`OnRobotAvailable`](https://github.com/equinor/flotilla/blob/23b00904d746a51691b5b44f662af1b5cb48389c/backend/api/EventHandlers/MissionEventHandler.cs#L109), a notification should be sent to the frontend so that an appropriate message can be shown to the user.
Currently we silently fail which might lead the user to believe that the robot is in its home.

**Describe the solution you'd like**
A signalR topic?

**How will this feature affect the current Threat Model?**
This will make the system more robust given that the user will be warned if a return to home mission could not be scheduled and the robot should be physically handled.

andchiind, 2023-12-27:
We could potentially have a generic signalR topic for alerts which sends a "FailureAlert" object containing a descriptive message as well as an alert name. We could then trigger frontend event handlers based on these alert names if more actions besides displaying the alert is needed.

andchiind, 2023-12-27:
We may also want to consider what the robot state should be at this point. I suppose they would become Available?

andchiind, 2023-12-27:
I think it should be up to the backend to update the robot state, so that the backend is consistent with the frontend. Currently the backend state becomes inconsistent I think, since the mission queue may remain frozen even after a failure. I think the safezone status of the robot should either be represented in the robot status, where we could also have an "inconsistent" or "needs help" status, or we need to expand the IsMissionQueueFrozen value to be more informative.

PR #1294 (2024-01-02): Add alert for failure to send robot to safe zone
Closes #1291
closed 2024-01-02 by andchiind