equinor/flotilla #1342: Run tests towards postgres database

aestene, 2024-01-18:
**Describe the new feature you would like to see**
The tests are currently running towards an in-memory SQLite database. From our experience this does not capture all issues which might arise in the Postgres database used in our environments.

**Describe the solution you'd like**
Investigate a solution where the tests may be run towards a proper postgres database. A tradeoff must be made in efficiency and speed. Testcontainers will be evaluated to see if it is suitable. We would like to have a solution for integration tests no matter what.

**How will this feature affect the current Threat Model?**
Improved due to better test coverage and quality.

Christdej, 2024-08-05:
Why Blocked? @aeshub

tsundvoll, 2025-01-27:
Done. We can now run test towards a postgresql database in a container

closed 2025-01-27 by tsundvoll