equinor/flotilla #1359: Robot has status offline if the broker restarts

prasm313, 2024-01-26:
**Describe the improvement you would like to see**
If the broker goes down and is restarted, the robots has "offline" as their status. If the robot is available or busy when the broker restarts this should be visible in Flotilla.

**How will this change existing functionality?**
Robots will not be stuck in offline. Will not be necessary to manually set status as Available through swagger. The users would not need to contact us if the broker has restarted.  

**How will this improvement affect the current Threat Model?**
N/A

Afonso-2403, 2024-02-07:
Could updating the robot status in [OnIsarRobotHeartbeat](https://github.com/equinor/flotilla/blob/116ae51d675578026bfcf204ed94a2ffb06cb124/backend/api/EventHandlers/IsarConnectionEventHandler.cs#L60) if the robot is not enabled solve this problem?
The only thing is that as of now the robot status is not contained in the heartbeat.

tsundvoll, 2024-04-29:
Blocked waiting on improved communication between ISAR and Flotilla. This includes some new states in ISAR. It also means we can rely on one of the messages instead of the combined status message.

tsundvoll, 2024-10-07:
Ready to be retested

tsundvoll, 2024-10-21:
Still an issue. Consider using the "heartbeat" to check regularly for a change in status.

PR #1781 (2024-10-30): Fix flotilla not reporting connection issues after broker goes down
Closes #1359 

Previously we were sending signalR messages that the robot had been updated with isarConnected being changed, but since the robotModel was null we were ignoring it in the frontend. This was occuring since we were not including the robot model in the database query we used when updating isarConnected. When I changed this to include the robot model we got an error since the ExecuteUpdateAsync function does not allow SQL JOINS (see https://learn.microsoft.com/en-us/ef/core/saving/execute-insert-update-delete#naviga [... cut]
closed 2024-10-30 by andchiind