equinor/flotilla #1366: Move installation filtering of robots

Eddasol, 2024-01-26:
**Describe the improvement you would like to see**
Currently robots are filtered by installation several places in the frontend. As installationcode varies between lower letter and capital letter, the installationcode needs to be made lower letter before comparison in the filters. This needs to be remembers every time a installation filter is made.

Suggest moving the filtering to one place. Preferably to the backend.
 
**How will this change existing functionality?**
Less prone to bugs and less repeated code

**How will this improvement affect the current Threat Model?**

UsamaEquinorAFK, 2024-02-05:
Made it lower case upon access from database https://github.com/equinor/flotilla/pull/1390

UsamaEquinorAFK, 2024-02-05:
[image]

Works

closed 2024-02-05 by UsamaEquinorAFK