equinor/flotilla #1383: Update Robot status during heartbeat handling

andchiind, 2024-01-31:
**Describe the bug**
Robots often get stuck in busy even though Isar reports both the missionStatus as being idle and robotStatus as being available. This is because we do not update the robot status in "OnIsarRobotInfo". I suggest we update the robot status at the end of this event handler, but I am open to suggestions as for why this is not the best idea.

**To Reproduce**
The robot gets stuck in busy when, for instance, Isar completes a mission as soon as it's received. This happens for instance when localisation is not required for the given Isar instance. Currently the workaround is to make isar sleep briefly after receiving such missions.

**Expected behavior**
If Isar reports itself as being available, then this should be relayed to the Flotilla data model in when looking at the heartbeat messages.

**Screenshots**
[image]
Also: image a picture where the Flotilla frontend says "Busy".

Christdej, 2024-08-12:
Waiting for new status solution in ISAR

andchiind, 2025-02-13:
Closed for now, can be opened again if we encounter the problem again.

closed 2025-02-13 by andchiind