equinor/flotilla #1386: Missions not starting when robot is ready

andchiind, 2024-02-02:
**Describe the bug**
Even though the robot is ready and there are no ongoing missions, the missions are not being started.

**To Reproduce**
Run isar_robot, but remove the sleeps in initiate_mission and initiate_step (this is a realistic test for robots that do not require localization). Then schedule a few missions, wait for them to finish and repeat. This bug will likely happen eventually.

**Expected behavior**
If the robot is available and there are missions in the queue, then these missions should be started. The problem likely lies in the way return to home missions work. In this case the robot was restarted at one point, which resulted in return to home being scheduled when it restarted.

**Screenshots**
[image]
[image]

aestene, 2024-05-25:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2024-07-25:
Not able to recreate anymore

closed 2024-07-25 by andchiind