equinor/flotilla #1387: Robot stuck in busy after missions complete quickly

andchiind, 2024-02-02:
**Describe the bug**
When missions are completed near instantly in ISAR, the state in Flotilla becomes inconsistent. It displays that the robot is busy and therefore does not start any new missions after the last completes. This happens even though the robot is Available and the mission status is Idle in ISAR. This is particularly problematic when it comes to Isar instances which ignore localization.

**To Reproduce**
Remove the sleeps in initiate_mission and initiate_step in isar_robot, then run any mission with a single task (such as localization, but using not-stepwise might also work for missions with more tasks).

**Expected behavior**
The hearbeat messages from Isar should either be able to correct the state, or the other event handlers in Flotilla need to be less reliant on events arriving in a specific order from ISAR.

**Screenshots**
[image]
[image]

UsamaEquinorAFK, 2024-02-19:
Unassigning myself becuase of pending discussion.

oysand, 2024-03-04:
When specifying the STEP_DURATION_IN_SECONDS in isar-robot below the ROBOT_STATUS_PUBLISH_INTERVAL in ISAR, the robot status was not sent to flotilla. When increased above this level the status was sent to flotilla

oysand, 2024-03-11:
This sleep makes it so that if the entire step completes within this time it does not send anything since the combined status is the same as the previous robot status

[image]

andchiind, 2024-06-11:
Closed until we re-encounter the issue. Suspected solution implemented