equinor/flotilla #1393: Error message card bug/improvement

prasm313, 2024-02-06:
**Describe the bug**
The error card does not adjust height when the error message is long.

Actions:
- Make cards with flexible height to make text fit
- Make the numbers fit better into the text
- Adjust the text alignment to the left
- Add translation to pressure error

**To Reproduce**
Set pressure threshold in initDB.cs to a high value.
LowerPressureWarningThreshold = 0.08f,

**Screenshots**

[image]

Pagla-Dasu, 2024-02-07:
If the issue still prevails, do assign me this, I would like to work on this.
Also the pull request guide for contributors are not working, so do mention the message I have to give for the pull request.

closed 2024-02-07 by prasm313

prasm313, 2024-02-07:
Hi @Pagla-Dasu! Thank you for your interest in contributing, but at the moment we don't have the code ready to be able to have open source contributions. Ref #1046. 

Regards, Petter

Pagla-Dasu, 2024-02-07:
Okay, Thank you.