equinor/flotilla #1412: Global Ordering for Mission Run

UsamaEquinorAFK, 2024-02-12:


oysand, 2024-02-12:
How do we handle editing of the mission ordering if a user wants to rekorder the queue?

UsamaEquinorAFK, 2024-02-12:
In the scope of this PR. I intend to introduce ordering and make logic for assigning lowest number to localization mission run. For that we should make the frontend queue interactable so that we can drag the queue items up and down.

UsamaEquinorAFK, 2024-02-12:
Holding this off for now to prioritize stable operations. Localization can be handled by adding a new field inside enum for MissionRunPriority.

aestene, 2024-05-03:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2025-02-13:
See https://github.com/equinor/flotilla/issues/1227#issuecomment-2656078422

Eddasol, 2025-06-10:
Do we need this?

andchiind, 2025-06-12:
> Do we need this?

If we only want to rely on the automatic mission scheduler instead of manually scheduling missions, then no. If we want to be able to rearrange missions that are already scheduled, then yes.

tsundvoll, 2025-06-16:
Not planning to re-order the queue, so this is not needed

closed 2025-06-16 by tsundvoll