equinor/flotilla #1428: Error message after changing robot facility

mrica-equinor, 2024-02-20:
**Describe the bug**
When we change a robot from one facility to another, the robot is still localized in the original facility. Isar-taurob will try to fetch the map for the area it was localized in and won't find it because it doesn't exist in the new facility.

This can be fixed by localizing the robot in an area in the new facility.

**Screenshots**
Porthos was moved from K-lab to JCA so then we got this error message: 

[image]

**Expected behavior**
The error message should contain more information about what is happening.

tsundvoll, 2024-02-26:
My understanding of this issue is to give better error message to the user of the reason, or to have an automatic re-localization or other fix in ISAR. We can work around it, therefore not prioritized for now.

aestene, 2024-05-03:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Christdej, 2024-08-12:
We expect different behavior due to rewrite of localization

closed 2024-08-12 by Christdej