equinor/flotilla #1442: Time used is not correct during the mission

anetteu, 2024-02-29:
**Describe the bug**
During a mission, time used is not correct for all missions. Typically, it shows 60-80 minutes instead of 0-20 minutes. I have not confirmed it, but suspect that there might be an hour offset. The time used is shown correctly for the mission in the history page. 

**To Reproduce**
Steps to reproduce the behavior: Start a mission and monitor the time used on the mission page during the mission. 

**Expected behavior**
The time used should be correct. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

Eddasol, 2024-03-11:
Suspect difference in time zone for missions received through API call and missions received through signal R

This PR ensures local time for variables fetched through API call. Might help understand the issue
https://github.com/equinor/flotilla/pull/1330

tsundvoll, 2024-03-11:
@aestene , do you have some input on the way to solve this?

andchiind, 2024-03-13:
The issue seems to be related to this: https://stackoverflow.com/questions/40205893/datetime-kind-set-to-unspecified-not-utc-upon-loading-from-database
When we start a mission, the datetime "kind" value is set to UTC, which is received at the frontend with the "Z" flag at the end. When we read the mission in from the database however, the "kind" is not set, causing it to be sent to the frontend without the flag.

andchiind, 2024-03-20:
See https://github.com/equinor/flotilla/pull/1476

tsundvoll, 2024-04-24:
If this is not formalized somewhere, I think we should always store a datetime as UTC, and only when we display it in the frontend will we change it to local time. Any thoughts on this strategy, @andchiind  and @Eddasol

PR #1471 (2024-03-19): Manage UTC/local time inconsistency
Closes #1442
closed 2024-06-13 by Eddasol