equinor/flotilla #1445: Improve assert for ReturnToHomeMissionIsStartedIfQueueIsEmptyWhenRobotBecomesAvailable

oysand, 2024-03-01:
**Describe the improvement you would like to see**
The test currently only assert that there is no ongoing mission, which does not take into account whether or not the previous completed mission run was a return to home mission.
```
            // Assert
            Thread.Sleep(1000);
            var ongoingMission = await _missionRunService.ReadAll(
                new MissionRunQueryStringParameters
                {
                    Statuses = [
                        MissionStatus.Ongoing
                    ],
                    OrderBy = "DesiredStartTime",
                    PageSize = 100
                });
            Assert.False(ongoingMission.Any());
        }
```

**How will this change existing functionality?**
More accurate tests

**How will this improvement affect the current Threat Model?**
N/A

aestene, 2024-05-03:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Eddasol, 2025-01-28:
Test is no longer found if described file

closed 2025-01-28 by Eddasol