equinor/flotilla #1447: Error message displayed to the user is not logged in the backend

tsundvoll, 2024-03-04:
**Describe the bug**
We got `Failed to start: 'Could not reach ISAR at http://localhost:3000'` popup in the frontend, but could not find a similar log in the backend pod.

**To Reproduce**
Steps to reproduce the behavior:
1. Trigger some failure
2. Verify that the frontend gives a error message in a banner
3. See that no error is logged in the backend

**Expected behavior**
The backend component should also log an error if an error is displayed in the frontend.

**Screenshots**

andchiind, 2024-04-02:
Hmm, our logger setup might have an issue then, since it should be getting logged in two places at least:

[image]

[image]

Christdej, 2024-08-05:
Need to send frontend logs to app insights

andchiind, 2024-08-13:
Could not be recreated, logs seemed to work fine.

closed 2024-08-13 by andchiind