equinor/flotilla #1450: We need a more robust way of resetting the current area to null

tsundvoll, 2024-03-05:
**Describe the bug**
Currently, the only way the current area is set back to null is when the "return to home" mission completes. We had a bug where if the "return to home" mission for some reason did not finish (in our case the backend actually crashed at this step, due to another bug, so it was a bit extreme), the current area was not set to null. The result of this is that the robot "thinks" it is still localized, even though no mission has been running.

**To Reproduce**
Steps to reproduce the behavior: see description of the bug.

**Expected behavior**
The current area should be set to null more robustly. For instance if there are no missions running and there are no pending missions in the queue; this could be a condition

**Screenshots**
If applicable, add screenshots to help explain your problem.

oysand, 2024-03-11:
Currently the current area is set to null when specific events happens. We would like to swap this so we would need to actively need to keep the current area from being set to null

Christdej, 2024-07-29:
Waiting for https://github.com/equinor/isar/issues/532

tsundvoll, 2024-09-02:
On hold for now. We can do this later as the whole localization procedure is likely to change anyways.

oysand, 2024-09-02:
Current work done here: https://github.com/oysand/flotilla/tree/robust-de-localize

andchiind, 2024-12-19:
No longer relevant as localisation has been removed from Flotilla

closed 2024-12-19 by andchiind