equinor/flotilla #1459: If robot battery is running low, stop the missions and send it home

mrica-equinor, 2024-03-08:
**Describe the new feature you would like to see**
We should estimate how much battery we need to run the robot back home before it reaches the baterry limit. For example, if the limit is 10% and from where we are it takes 5% to go back home, when the robot reaches 15% battery it should go home. 

Since we don't have a way of estimating the battery necessary to go back home at the moment, I believe we could use the lower baterry limit until another solution is found. 

**Describe the solution you'd like**
If the robot reaches the lower battery limit, the ongoing mission should be stoped, none of the missions in the qeue are executed and the return to home mission starts.

**How will this feature affect the current Threat Model?**

tsundvoll, 2024-03-11:
We also need a similar thing for pressure where relevant

prasm313, 2024-06-04:
Working on [#1491](https://github.com/equinor/flotilla/issues/1491). I want to add pre-check (pressure and battery status) between each mission run, is that good enough for solving this issue?

PR #1684 (2024-07-25): Make robot go home when battery or pressure levels are outside safe limits
Closes #1459