equinor/flotilla #1463: Adding missions to the queue before localization missions is finished

mrica-equinor, 2024-03-11:
**Describe the bug**
When scheduling additional missions to the queue while there is an ongoing localization mission, the mission is not added. This is due to the fact that for the mission to be added to the queue, it has to be in the same deck as the robot is localized and since the localization mission hans't finished, the robot doesn't have a deck yet.

Suggested solution is to have a check when adding a new mission to the queue for ongoing localization missions and if the mission added is in the same deck as the ongoing localization mission, then it should be added in the queue.

**To Reproduce**
Add mission to the queue before the localization mission updated the robot current area.

**Expected behavior**
As long as the mission to be added is in the same deck as the ongoing localization mission or in the same deck as the robot current area, it should be added to the queue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

andchiind, 2024-03-25:
Alternate solution could be to set the deck of the robot as soon as we queue the localisation mission, and instead we can perform cleanup of the queue if the localisation mission fails. We anyways will not start other missions until the localisation mission finishes.

aestene, 2024-05-25:
This issue has automatically been marked as stale as there has been no activity for 60 days.

mrica-equinor, 2024-08-07:
This issue has changed. Currently it is possible to add the mission to the queue but the user still sees a popup confirmation that the robot is in the default localization pose for as long as the robot current area is null. 

On placebot this isn't an issue because the first task of the mission completes fast enough that the current area will be set almost immediately. But with a real robot, the first task might take several minutes. 

This issue will be therefore handled by changing the localization confirmation dialog to check if there are ongoing localization missions intead of relying on the robot having current area set to null.

PR #1718 (2024-08-13): Fix dialogs during ongoing localization mission
Closes #1463
closed 2024-08-13 by mrica-equinor