equinor/flotilla #1472: Improve enum MissionRunPriority

mrica-equinor, 2024-03-18:
**Describe the improvement you would like to see**
Currently the enum MissionRunPriority is used as a way to both filter which missions in the queue should be prioritized and the kind of mission they are. Instead, the enum should be changed to MissionRunType since this is a more accurate definition for the missions. 

Mission types should be:
- Normal
- Localization
- Emergency
- Move robot arm (see issue #1466 )
- ...

**How will this change existing functionality?**
This will make it easier to avoid scenarios where the return to home mission is triggered, for example. We can check that if the mission run was a move robot arm (which is a mission with no tasks other than move the robot arm), the return to home mission shouldn't be triggered.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

aestene, 2024-05-18:
This issue has automatically been marked as stale as there has been no activity for 60 days.

mrica-equinor, 2024-07-17:
Close this since it has been fixed

closed 2024-07-17 by mrica-equinor