equinor/flotilla #1517: Robot location does not need to be confirmed when starting echo missions

anetteu, 2024-04-06:
**Describe the bug**
Start an echo mission in flotilla. The pop-up to confirm the location of the robot is missing. 

**To Reproduce**
Steps to reproduce the behavior: 
Start an echo mission in flotilla. The pop-up to confirm the location of the robot is missing. 

**Expected behavior**
The pop-up to confirm the location of the robot should appear

**Screenshots**
If applicable, add screenshots to help explain your problem.

Eddasol, 2024-04-08:
This is cause we don't have access to area for echo missions. Awaiting update in Echo mission planner

tsundvoll, 2024-04-29:
Solving https://github.com/equinor/robotics-infrastructure/issues/487 will un-block / solve this issue.

tsundvoll, 2025-01-27:
We are considering to move localization functionality from Flotilla to ISAR, so this might not be relevant anymore. Closing, and we can reopen if we find it necessary

closed 2025-01-27 by tsundvoll