equinor/flotilla #1518: Add picture of robot at parking spot to pop-up for confirmation of the robot position?

anetteu, 2024-04-06:
Can be added in addition to the map. Can maybe be easier to recognize and compare?

anetteu, 2024-04-08:
This suggestion is supported by user. Replace map with picture and description of location (map and N/S/E/W). 

This issue is also related to #1011. The map appearing will be a different one than the map relevant for the mission if the home position is on a different map. There is no information about where that is if the user does not recognize the map.

oysand, 2024-04-10:
First step is to remove the map in the confirmation and have a short clear descriptive text

mariannewiggen, 2024-05-10:
Suggestion for a pop-up to confirm location. This suggestion can be used if the location is clearly marked (with e.g. an arrow) to explain the direction and position of the robot. Can add direction or something else to make it clearer.

[image]

prasm313, 2024-05-16:
> Do we have any pictures of the robot at the parking spots?

We have some pictures in Robotparking folder in Sharepoint, but this should be given as an action on the next trip(s).

andchiind, 2025-02-14:
Closing this as we now use docking stations for localization.

closed 2025-02-14 by andchiind