equinor/flotilla #1522: Conflicting decks warning does not show when scheduling echo missions

anetteu, 2024-04-07:
**Describe the bug**
When scheduling an echo mission, using the `+ Add predefined Echo mission`, there are no warning if the mission scheduling fails because it is on a different deck than the robot

**To Reproduce**
Steps to reproduce the behavior:
Try to schedule an echo mission, using the `+ Add predefined Echo mission` button, that is on a different deck than the robot. The mission will be canceled, but there are no warning or explanation.

**Expected behavior**
A clear and concise description of what you expected to happen.
If the same mission is scheduled from the menu under `Predefined missions`, a warning about conflicting decks will appear. This warning should also appear when the mission is scheduled using the `+ Add predefined Echo mission` button.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Eddasol, 2024-04-08:
This is cause we don't have access to area for echo missions. Awaiting update in Echo mission planner

tsundvoll, 2024-04-29:
Solving https://github.com/equinor/robotics-infrastructure/issues/487 will un-block / solve this issue.

tsundvoll, 2025-01-27:
Ready to implement, due to a change in how we fetch areas from STID