equinor/flotilla #1523: Update message in robot-is-blocked banner

anetteu, 2024-04-07:
**Describe the improvement you would like to see**
When the robot stops for an obstacle in its path, a red banner with the following message appears: `Robot is blocked and cannot perform tasks`. I suggest to change this to an information banner (other color than red?) and edit the message to something like: `Robot is blocked by obstacle and is replanning its route`.

**How will this change existing functionality?**
No change, just more accurate information

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

Eddasol, 2024-04-08:
Note that not all robot can re-plan so might need a more general message

andchiind, 2024-07-25:
> Note that not all robot can re-plan so might need a more general message

I suppose we should ideally not report blocked to Flotilla from ISAR until it has failed to replan

betaniat, 2024-08-05:
Close, fixed by linked issue.

closed 2024-08-05 by betaniat