equinor/flotilla #1526: Operator fails to delete mission from queue on iPhone

anetteu, 2024-04-07:
**Describe the bug**
Configuration: Operator user access, iPhone, safari 

Tried to delete mission from the queue many times. The operator confirmed the deletion of the mission, but did not succeed. The following error banner appeared: 

[image]

The mentioned missions were removed from the queue successfully with the configuration dev access, iPad, safari. 

**To Reproduce**
Steps to reproduce the behavior:
Configuration: Operator user access, iPhone, safari. Queue all mission on a deck. Then, try to remove a mission from the queue. 

**Expected behavior**
The mission should be removed from the queue

**Screenshots**
If applicable, add screenshots to help explain your problem.

tsundvoll, 2024-04-08:
Need a quick discussion about what should happen when a mission is deleted / removed, and which access should be required.

**Maybe this action should be possible to do with user access**. Specifically deleting / removing missions from the "Pending" missions can have a separate endpoint?

oysand, 2024-04-08:
This should allow operators to remove mission runs: #1532

closed 2024-04-09 by oysand