equinor/flotilla #1527: Front page does not update on iPhone

anetteu, 2024-04-07:
**Describe the bug**
Configuration: iPhone, webpage ran in safari saved on Home Screen, operator user access. 

The front page does not update to reflect to current situation on active missions etc. Did not succeed to manually refresh the page without exiting safari and open it again. 

**To Reproduce**
See above

**Expected behavior**
The front page should show updated status without the need for a restart of the app. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

tsundvoll, 2024-04-08:
Figure out if we are supporting Safari; if this is a specific Safari issue; or if it is a user access issue or something else is missing

mrica-equinor, 2024-04-10:
I believe this is related to #1535 and the solution is the same for both.

tsundvoll, 2024-05-13:
Possibly related to other refresh-issues. Go over the SignalR configuration to see if it is related to how this is set up.

andchiind, 2024-07-08:
Moving it to TODO until we have verified that it is still an issue.

Christdej, 2024-08-12:
Might be solved, needs to be tested

tsundvoll, 2024-09-02:
We need to define the environment where the app (Flotilla) is supposed to work.

Eddasol, 2025-01-28:
Not able to recreate

closed 2025-01-28 by Eddasol