equinor/flotilla #1528: Consider rewording of "safe zone"

anetteu, 2024-04-07:
**Describe the improvement you would like to see**
Consider renaming the button `Send robot to safe zone` with something like `Park all robots (outside escape route)` or similar. This to avoid any confusion about what a `safe zone` is. It this context, a `safe zone` can still be an ATEX sone, but it will be out of the way for escape routes and main walkways on the map. 

**How will this change existing functionality?**
No change in functionality, just rewording.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

prasm313, 2024-04-15:
I agree with your arguments and really like "Park all robots". As UX-designer has mentioned earlier, a button should not have more than three words. Tried to think of other wordings: "Move out of main path", "Clear walkway", "Relocate robots", but they are not as descriptive.

prasm313, 2024-04-17:
Have talked with UX-designer. They want to have input from users before changing the wording on the button.

anetteu, 2024-06-05:
What about something like "Send robots home"? The current wording is five words, how important is it to have only three? Do we need to wait for the optimal wording and user input, or can we change it and then change it again after we have found the optimal wording?

Eddasol, 2025-01-28:
Already rewritten from "safe zone" to "dock"

closed 2025-01-28 by Eddasol