equinor/flotilla #1537: Frontend screen sometimes goes white when scheduling multiple missions at once

oysand, 2024-04-09:
**Describe the bug**
Sometimes when scheduling all the missions on the mezzanine deck at Huldra in dev the screen is goes completely white. A refresh "fixes" the problem.

**To Reproduce**
Steps to reproduce the behavior:
Queue all the missions on the mezzanine deck at Huldra multiple times.

**Expected behavior**
Screen should not be blank

**Screenshots**
[image]

oysand, 2024-04-09:
Experienced the bug with updated mission topic as well with the missionRunContext

mrica-equinor, 2024-04-11:
I haven't been able to reproduce this bug so we might consider waiting on this to see if it keeps happening

oysand, 2024-04-18:
New instance seen in staging: 
[image]
Occured after opening staging and just leaving it on the side in a tab for approximatly 5-10 minutes. When revisiting the tab it was all white and had error on the robot status card

oysand, 2024-04-18:
New occurance: 
[image]

mrica-equinor, 2024-04-18:
[image]

[image]

Eddasol, 2024-06-12:
When adding this to signalRContext.tsx
[image]

the console.log message appears at the same time as these kind of "screen goes white"-events
[image]

Seems that the issue is caused by signalRMessages sometimes being empty

Eddasol, 2024-06-14:
Frontend part of this issue is solved by adding error handling in #1617. Backend issue and root cause is handled by #1604

oysand, 2024-06-14:
Reappered on robot status card
[image]

closed 2025-01-28 by Eddasol