equinor/flotilla #1544: Add dialogs when pressing the Safe Zone button

mrica-equinor, 2024-04-10:
**Describe the improvement you would like to see**
Currently the Figma sketch has dialogs when pressing the send robots to Safe Zone button.

**How will this change existing functionality?**
Functionality should remain the same.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

mrica-equinor, 2024-04-16:
[image]

Text:

Sende robot til sikker sone

Du er i ferd med å sende robotene til en sikker sone og dermed stoppe alle pågående oppdrag. Robotene vil fortsette på oppdragene når du bekrefter at det er trygt a fortsette.  

Er du sikker på at du vil sende alle robotene til en sikker sone?

haakonsf, 2024-05-16:
https://github.com/equinor/flotilla/commit/0caeae6b7c54f79cde0e6982619517916a6bab3d closes this issue

closed 2024-05-16 by haakonsf