equinor/flotilla #1547: Should we have a remove all missions from the mission queue button?

mrica-equinor, 2024-04-11:
**Describe the improvement you would like to see**
Add remove all missions button next to the mission queue so if we want to remove the missions, we don't have to do it one by one.

There is already code on the backend that clear all the queue if the localization fails which can probably be re-used on the new created endpoint.

**How will this change existing functionality?**
No changes.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

prasm313, 2024-06-03:
Could potentially be solved by #1599

andchiind, 2024-07-25:
This could also potentially be a button in the Robot page, if we intend to keep it.

closed 2025-04-01 by mrica-equinor