equinor/flotilla #1556: Only allow selecting an available robot when scheduling a mission

anetteu, 2024-04-15:
**Describe the improvement you would like to see**
When scheduling a mission, a pop up with a drop down appears to select a robot. Currently, it is possible to select robots that are not available as they are not on the correct deck. I suggest to include all robots on the asset in the drop down menu, but only make the ones available for that particular mission selectable.

**How will this change existing functionality?**
Provide more information and context to the robots when selecting them for a mission.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

andchiind, 2024-04-17:
This might be difficult to do currently, as robots do not have an area value until they are localised, and they are only localised when running missions.

Eddasol, 2024-04-18:
This can potentially be confusing if the user is expecting to see/select a specific robot and then it's just missing from the list without providing information about why. Isn't it better to allow the user to select wrong and instead give them a dialog about why its not possible (Conflicting deck dialog)? Alternatively, (not sure if possible) disable none-available robots in the drop-down and use tooltip to provide information if a user still tries to select it.

anetteu, 2024-04-18:
Yes, disable non-available robots was what I was thinking, the error is already there. For assets with several decks and robots the user has to try and fail to select the correct robot. 

It can also be an option to display the location of each robot on the robot cards? 

It is related to #1011

mrica-equinor, 2024-07-17:
Filter robots that are available or busy

PR #1677 (2024-07-19): Filter robots by available or busy status
Closes #1556
closed 2024-07-19 by betaniat