equinor/flotilla #1560: Implement notification feature

prasm313, 2024-04-17:
**Describe the new feature you would like to see**
Implement the notification feature so that the users have an overview of the status of the missions and robots.

**Describe the solution you'd like**
Figma sketch:

[image]

**How will this feature affect the current Threat Model?**
N/A

anetteu, 2024-04-17:
Consider #1087 when implementing this

anetteu, 2024-04-17:
Suggest to add button to the battery and pressure warnings: `Stop missions and drive home`.

prasm313, 2024-06-06:
If the alert is regarding a robot, the user should be sent to the robot page when clicking the alert. Clicking alerts related to missions should send the user to missionRun page.

prasm313, 2024-06-25:
The work has started in PR #1645.

haakonsf, 2024-08-29:
Frontend display implemented with no functional change to alert generation or banner alerts. See comment in #1592 for more details.

tsundvoll, 2024-09-02:
Considered done for now. Create new issue if adjustments are wanted

closed 2024-09-02 by tsundvoll