equinor/flotilla #1567: Cannot handle tags that go over several decks

Eddasol, 2024-04-22:
**Describe the bug**
For echo missions:
Tags can only be registered for one area and if a tag goes over several decks it can cause issues.

Assume a tag tag1 that is on both deckA and deckB but is registered as being on deckA. If we want to inspect a part that is on deckB there will be issues. If tag1 is one of several tags (assuming normal tags on just one deck) then the mission will fail to be scheduled. If tag1 is the only one then the mission will be added to the inspection plan for deckA. 

**To Reproduce**
Schedule mission with tags that are registered as being on a different deck than the belonging robot/inspection pose

**Expected behavior**
Should have a way to handle tags that spans more than one deck

**Screenshots**

andchiind, 2024-04-26:
Could a solution be to remove line 288 in MissionSchedulingController, and instead just put a LogWarning?

[image]

andchiind, 2024-04-26:
I have made a PR which prevents an error being thrown when one or more tag is in a different deck, but this does not address the case where more tags are in a different deck/area than the one where we are actually running the mission on the robot. This is perhaps best solved within Echo, so I am moving this issue to blocked for now.

tsundvoll, 2025-03-03:
Now we only care about the position where the robot is standing, not the position of what you are looking at. We check this towards the newly created concept of Polygon.

Therefore we don't check the area of the tagged equipment anymore.

- [ ] Check if we are still using it

tsundvoll, 2025-03-03:
Now we only care about the position where the robot is standing, not the position of what you are looking at. We check this towards the newly created concept of Polygon.

Therefore we don't check the area of the tagged equipment anymore.

We are soon to remove the usage of this, so closing as not relevant anymore

closed 2025-03-03 by tsundvoll