equinor/flotilla #1578: Rewrite map service

eivindsjovold, 2024-05-02:
**Describe the improvement you would like to see**
Currently the map service is dependant on loading png images from an azure blob service. Rewrite this to be more flexible for non-Equinorian users. The service then approxes position from pixels in image.
- [  ] 

**How will this change existing functionality?**
Easier to run for external companies, less coupling in program

**How will this improvement affect the current Threat Model?**

Christdej, 2025-01-15:
This can also make it possible to zoom and move the map to make it more user friendly

mrica-equinor, 2025-07-07:
Moving this to blocked since we are waiting for Pointilla to be ready for integration with Fencilla

tsundvoll, 2025-09-22:
We are changing our approach towards the map now anyways and are not relying on blob storage anymore.

See new issue that replaces this: #2408