equinor/flotilla #1582: Implement the Facility interface and microservice

aestene, 2024-05-02:
**Describe the new feature you would like to see**
A `FacilityController` combined with a `FacilityService` should replace existing functionality talking to Echo and other relevant third-party applications. The goal is to have a solution in Flotilla which can talk to an external microservice through configuration instead of being tie to Echo. This will enable others to use Flotilla more easily without having a specific requirement for Echo.

**Describe the solution you'd like**
The existing `EchoController` should be replace by a `FacilityController` which handles requests from the frontend which are redirected to Echo. It will then use a `FacilityService` which makes calls to an external microservice through an HTTP request. The host and port of the microservice should be configurable. 

The microservice should be hosted as a docker container with a default dummy implementation available in the Flotilla repository. For Equinor a different repo should be made to create the Echo facing microservice. The system as a whole should be designed such that a different operator could create their own microservice and use that with the system instead of the dummy or Echo. 

Suggested solution is to use botnet for the microservice. 

**How will this feature affect the current Threat Model?**
This will introduce new HTTP requests which require authentication on the microservice side.

aestene, 2024-05-02:
This will replace #1580

andchiind, 2024-05-15:
👀👀 https://camunda.com/blog/2022/09/seven-best-programming-languages-for-microservices/#haskelllanguage

closed 2025-03-10 by andchiind