equinor/flotilla #1592: Improve pre-mission checks and alerts

prasm313, 2024-05-15:
**Describe the improvement you would like to see**
- [ ] Should implement all checks (offline, pressure, battery) in one function (not all places where a mission can be scheduled)
  - [ ] Should not be able to schedule a mission when the robot has connection issues
  - [ ] When a robot is running on Deck1 you should not be able to schedule a mission on Deck2
- [ ] Should remove possibility to rerun a "Return to home" and "Localization" mission. (today you get a alert banner "Failed to rerun mission")
- [ ] ~~Should be consistent with pressure warnings: Are we using pop-ups or red warning banners.~~
- [ ] Refreshing the page removes "Request error" alert banners. Does not remove "Failed" banners. Banners should be available, even when refreshing, for at least 10 minutes. 
- [ ] Implement alert banner for aborted missions
- [ ] Should implement notification for alerts. This gives a better overview for user, overwritten alerts will not disappear and alerts will not disappear when refreshing. [#1560](https://github.com/equinor/flotilla/issues/1560)

**How will this change existing functionality?**
Will make Flotilla easier to navigate for the users.

**How will this improvement affect the current Threat Model?**
N/A

haakonsf, 2024-08-27:
Edited text to reflect current identified gaps for a well working alerts feature. #1713 provides the implementation of a notifications drop down menu, but does not change alerts behavior. With this alerts are generated twice in order to not remove any current functionality. This is temporary until a discussion on desired notification behavior and what alerts to display when is had.

haakonsf, 2024-09-09:
From a previous UX discussion:
[image]
[image]