equinor/flotilla #1596: Error handling for when a robot does not reach it's safe zone

haakonsf, 2024-05-15:
**Describe the improvement you would like to see**
Ref #1544 : It should be made more clear for the user when a robot fails to go to it's safe zone. Currently a banner is displayed stating that the mission has failed, but the status of the robot is "Safe Zone". This could potentially cause confusion. Nothing other than the banner displaying happens when the robot fails to go to a safe zone. 

- [ ] Consider adding another status to show it has not returned to safe zone
- [ ] Consider adding error handling for situations where the robot fails to go to safe zone

**How will this change existing functionality?**
Improve user understanding and build on the go to safe zone feature.

**How will this improvement affect the current Threat Model?**

tsundvoll, 2024-05-27:
I would like this functionality to reside with ISAR.

Safe Zone ("In safe zone" or / "sideline") could be a separate state in ISAR; and it will only transition here if it has verified that it is within "Sideline" borders.

mrica-equinor, 2025-07-07:
Outdated issue, replaced by https://github.com/orgs/equinor/projects/85/views/4?pane=issue&itemId=118753428&issue=equinor%7Cflotilla%7C2271