equinor/flotilla #1599: Options when clicking stop button

prasm313, 2024-05-30:
**Describe the new feature you would like to see**
When clicking the stop button on a mission, the operator should get the option whether to stop the robot completely or to run a Return to home mission.

**Describe the solution you'd like**
Pop-up with text and buttons similar to:

[image]

**How will this feature affect the current Threat Model?**
N/A

prasm313, 2024-06-03:
I will have a chat with UX designer

anetteu, 2024-06-04:
Some thoughts about this:

**Considering safety, I think the following behavior shall be considered:**
1. When the stop button is pressed, the robot stops immediately
2. The robot shall not move before an action for the robot to do so is taken

**Furthermore, I suggest that:** 
1. There is only one stop button, which is connected to the robot and not to any mission. 
3. This stop button should be on a fixed position on the screen on all pages. It should not be necessary to navigate to access the stop button.  #901 (closed but not implemented?)
4. The robot shall stop immediately after pressing the button, without the need to answer any pop-up etc. 
5. After the stop button is pressed, and the robot has stop any movement, the user can be given some options, for example:

  - Continue 
  - Cancel current mission and continue with queued missions if any
  - Cancel all missions and go home
  - Cancel all missions and remain at the same position 
  - etc. 

Which should comply with the scenarios described above
 
6. Pause button can be removed. 

**This will address what I see as challenges with the solution we have today:**
1. The stop and pause buttons are not available on all pages and not in a fixed position. Navigation and scrolling is necessary to access the stop button. This can delay the operator in stopping the robot.
2. The robot does not stop until the operator has answered the pop-up. This delays the stop off the robot. Furthermore, if it is important to stop the robot fast, the operator will most likely just click and not actually read the pop-up.
4. The stop button does only stop the mission and not the robot. If there are missions in the queue, the robot will start driving again soon after the stop button is pressed even if there are no actions taken by operator.

mrica-equinor, 2025-07-07:
Closing until we get feedback that this is needed as an improvement