equinor/flotilla #1650: Needs to confirm localization even if the robot is already running a mission on the same deck

Eddasol, 2024-06-26:
**Describe the bug**
Needs to confirm localization even if the robot is already running a mission on the same deck. Localization shouldn´t be necessary when the robot is currently running a mission

**To Reproduce**
Steps to reproduce the behavior:
1. Schedule a mission on some deck
2. Schedule an additional mission on the same deck
3. Note the popup when scheduling the second mission

**Expected behavior**
There shouldn´t be a localization confirmation popup is the robot is already localized and running a mission

**Screenshots**

andchiind, 2024-07-08:
Related to this [#1655](https://github.com/equinor/flotilla/issues/1655)

andchiind, 2024-07-08:
Closing it as fixing [#1655](https://github.com/equinor/flotilla/issues/1655) should also fix this

closed 2024-07-08 by andchiind