equinor/flotilla #1657: Say failed to localize if canceling a mission

Eddasol, 2024-07-01:
**Describe the bug**
When canceling a missions there becomes a banner saying "Failed localization mission".

There are two issues with this:
1. As canceling doesn't have anything with localization to do (and the localization step was completed and successful), this is a misleading message and needs to be improved
2. Canceling a mission shouldn't lead to a failure message as nothing actually failed, it was just purposely cancelled

**How to reproduce**
1. Start a mission
2. Cancel the mission (after the first step such that the localization should be good)
[image]

3. Note banner popping up
[image]

mrica-equinor, 2024-07-05:
This issue is a consequence of this: 
https://github.com/orgs/equinor/projects/85?pane=issue&itemId=69123918

andchiind, 2024-07-08:
Closed as duplicate. See above comment

closed 2024-07-08 by andchiind