equinor/flotilla #1661: Figure out why checking robot.model.type fixes it being null

oysand, 2024-07-01:
**Describe the bug**
When isar finished the mission the screen turned white and it complained that type was null. After reproducing it locally the following check #1659  was added during debugging and then the problem dissapeared. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to the dev database locally.
2. Revert #1659 
3. Run a mission and observe what happens when the mission completes 

**Expected behavior**
Would like some other fix than just doing an if check

**Screenshots**
If applicable, add screenshots to help explain your problem.

oysand, 2024-07-01:
Connected to #1537

oysand, 2024-07-01:
The if check never true, but checking the value makes the system investigate the value and stops it from being treated as null further down the line.

aeshub, 2025-11-05:
Old and too many changes since discovery to reproduce