equinor/flotilla #1686: MissionIsCreatedInArea test is nondeterministic

andchiind, 2024-07-23:
**Describe the bug**
Sometimes the test fails other times it passes.

**To Reproduce**
Run the test several times locally.

**Expected behavior**
It should behave the same each time.

**Screenshots**
The screenshots were taken without any code changes being made inbetween them:
[image]

[image]

andchiind, 2024-08-01:
AddDuplicateAreasToDb_Fails has also been observed to fail, but only when running all tests together. I have also only seen it fail when running the tests for the first time after a code change. I do not have the error saved, but it complained about creating a Deck.

Christdej, 2024-08-05:
Hard to reproduce

andchiind, 2024-08-07:
Another example: 

[image]

tsundvoll, 2024-09-02:
Seems to be related to multiple event handlers running the same context. Consider if possible to run with separate context for each event.