equinor/flotilla #1707: Show in the frontend whether the robot is probably not localised

andchiind, 2024-08-06:
**Describe the new feature you would like to see**
If the robot is not running a mission and it is available in ISAR, the status of the robot should indicate whether we believe it returned to docking position. When we hover over this it would also be nice to provide info to the operator that they need to teleoperate it back to its docking location.

**Describe the solution you'd like**
I suggest we make a new robotFlotilla status which is diplayed instead of robotStatus.Available if it is set to something other that robotFlotillaStatus.Normal. Then the operator knows that no mission is running and that a mission can theoretically be run, but the robot is potentially not at home since the last return to home mission did not succeed. This should appear as a mild warning, not as an aggressive icon in my opinion.

**How will this feature affect the current Threat Model?**
N/A

mrica-equinor, 2024-08-13:
I don't agree with creating a new robot status, I think it should be based on wheather or not the robot current area is null

andchiind, 2024-08-29:
> I don't agree with creating a new robot status, I think it should be based on wheather or not the robot current area is null

This is not enough information to know whether it returned to its docking station though.

Eddasol, 2025-01-28:
Robot no longer gets delocalized when failing return to home. Remaining localization issues is handled by https://github.com/equinor/isar/issues/665

closed 2025-01-28 by Eddasol