equinor/flotilla #1720: Remove hardcoding of pressure limits

Eddasol, 2024-08-14:
**Describe the improvement you would like to see**
Rewrite pressureSection to remove hardcoding of the pressure limits
[image]

Also suggest changing such that the pressureTable isn't only for taurobInspector, but for all robot types with pressure

**How will this change existing functionality?**

**How will this improvement affect the current Threat Model?**

anetteu, 2025-04-30:
Closing issue as this is fixed in previous PR

anetteu, 2025-05-14:
Suggest to keep table as it is - it can most likely be removed within short time as this information is not needed in flotilla.

tsundvoll, 2025-05-26:
Can be removed now, we don't need to show this in Flotilla