equinor/flotilla #1737: Write and host Microservice to have a GetEchoMap service

tsundvoll, 2024-09-16:
**Describe the new feature you would like to see**
A server that exposes an API to get 2D maps from ECHO

**Describe the solution you'd like**
Parameters (roughtly):
* Plant name
* Area name
* x_min, x_max
* y_min, y_max
* z_height

Outputs 2D map with this crop.

We could then call this API for every new mission. To find the x- and y- -min and -max, we could iterate over all the tasks in the mission.

Points to solve:
* Caching of images (either on Flotilla side or on the API side)
* Access control to the images (and the sources for generating the images, the 3D models)

**How will this feature affect the current Threat Model?**
Easier to have this as a separate module for other users.

andchiind, 2025-02-13:
See https://github.com/equinor/flotilla/issues/1885

andchiind, 2025-09-16:
This has been replaced by a pointilla solution

closed 2025-09-16 by andchiind