equinor/flotilla #1739: GetAvailableMissions is too slow if many missions

Eddasol, 2024-09-18:
**Describe the bug**
If clicking the "Add predefined missions" button on an asset with several longer missions, the loading is unreasonable long. This is if using echoMissionLoader as selected missionLoader.

This is probably because the missionLoader is iterating through all missions and calling ProcessEchoMissions which is iterating through all tasks and inspections.  

**To Reproduce**
Steps to reproduce the behavior:
1. Select an asset with several longer missions
2. Click the "Add predefined missions"-button

**Expected behavior**
This shouldn't take several minutes. Tested on one asset with 75missions, and the fetching took 2,5 minutes.

**Screenshots**

Eddasol, 2024-09-19:
Bug was introduced in this commit: https://github.com/equinor/flotilla/commit/54ba48c323403d1e3122b802bd79521354b3d06b

Christdej, 2024-09-30:
One suggestion is too make Area optional on MissionDefinition and only load Area once someone is trying to start the mission.

tsundvoll, 2024-10-21:
The plan is to get the area information from the mission planner (Echo for example). Waiting on implementation for this

tsundvoll, 2025-01-27:
Maybe we don't need to check area for each task. Related to the new concept of "Navigation Area" instead of Area in STID/Echo/Equinor terms.

The Areas are overlapping and not well defined. Many tags does not have an area. It might be better to not rely on area.

Create an epic for this new concept that can replace Area (-> Navigation Area)

Eddasol, 2025-03-20:
This has significantly improved

closed 2025-03-20 by Eddasol