equinor/flotilla #1757: Wrongly giving recharging as status

Eddasol, 2024-10-10:
**Describe the bug**
When I set battery threshold to 94% and drove a mission, the status changed to Recharging

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**

**Screenshots**
From robot card while the robot was running a mission
[image]

tsundvoll, 2024-10-14:
Related to messages coming from SignalR. Issue to fix the root cause: https://github.com/equinor/flotilla/issues/1761

Eddasol, 2024-12-12:
Can use robotState introduced in this PR to fix:
https://github.com/equinor/flotilla/issues/1757

Eddasol, 2025-01-28:
Solved, will no longer claim recharging when heading to dock. Remaining issues is handled by [#1877](https://github.com/equinor/flotilla/issues/1877)

closed 2025-01-28 by Eddasol