equinor/flotilla #1760: Custom mission don't currently work

andchiind, 2024-10-14:
**Describe the bug**
When we try to schedule a custom mission the mission loader assumes that all missions are echo missions and tres to fetch the mission from echo using the source ID. This causes an error as the ID is not an integer for custom missions, but rather the hash of the tasks of the mission. The problem can be traced to the "GetMissionById" function in "EchoAndCustomMissionLoader.cs", which is expected to handle echo and custom missions, but which currently always assumes that we are using echo missions.

**To Reproduce**
There are some example custom missions in Dev. Running these will cause an exception the echo service.

**Expected behavior**
We currently store custom missions directly in the Source model. When scheduling a mission we can check whether this value is set, and, if so, use the SourceService to read in the mission. As a further improvement, we could add an enum to the Source model which can specify the mission loader to use for that mission.

**Screenshots**

tsundvoll, 2024-10-14:
How can we make sure that both models are up to date (Echo Mission and Custom Mission)?

andchiind, 2024-10-21:
> How can we make sure that both models are up to date (Echo Mission and Custom Mission)?

I am not sure I understand this question. We do not need to edit any models to solve this issue.

closed 2024-11-25 by tsundvoll